chenxiaolong/Custota #5: caddy: Error: unknown shorthand flag: 'a' in -access-log

archandanime, 2023-08-16:
Hi, thank you for your excellent work with very detailed documentation!

Yesterday I bought a new Pixel 7a(lynx) and and immediately tried Cusota. I successfully created `.csig` and `json` files but when trying to run caddy server I encountered:
```
> sudo caddy file-server -access-log -listen :8080
Error: unknown shorthand flag: 'a' in -access-log
```
My caddy version is `v2.6.4` on Arch Linux
I have never used caddy and really have no clue.
I also have suggestions:
- Make the Cusota app hint typing port on the URL(eg. URL:port), it might be not very obvious for some users type in the port.
- Add details about how often Cusota app automatically checks for update.

chenxiaolong, 2023-08-16:
Glad you're liking Custota!

> ```
> sudo caddy file-server -access-log -listen :8080
> Error: unknown shorthand flag: 'a' in -access-log
> ```
> My caddy version is `v2.6.4` on Arch Linux

Oops, it looks like I was accidentally using Caddy 2.5.2 from Fedora's repos (unfortunately unmaintained package) instead of my own 2.7.3 package. It looks like 2.6.0 and newer use `--access-log` instead of `-access-log`. I'll update the documentation.

> - Make the Cusota app hint typing port on the URL(eg. URL:port), it might be not very obvious for some users type in the port.

Good idea. I'll do that.

> - Add details about how often Cusota app automatically checks for update.

Sure. I'll add something to the docs about this. It checks at most once every 6 hours. It's not a fixed timer though. It uses Android's `JobScheduler` mechanism so it can run sooner or later depending on what Android thinks is the most battery efficient.

(The unmetered network and low battery options also entirely handled by `JobScheduler`. Custota never wakes up the CPU just to determine if an update check is necessary. Android only starts Custota when the conditions are met.)

> - Add details about "sufficient battery level" by Custota app.

This one depends on the ROM and there's unfortunately no way to query what the exact value is. Android's documentation only says it might match when the low battery popup shows up (but not guaranteed). I'll also include this in the docs.

archandanime, 2023-08-17:
> Oops, it looks like I was accidentally using Caddy 2.5.2 from Fedora's repos (unfortunately unmaintained package) instead of my own 2.7.3 package. It looks like 2.6.0 and newer use --access-log instead of -access-log. I'll update the documentation.

It works!
-----
I think it's a good idea to also have `--root /path-to-Cusota-directory/` so people can run the command from anywhere, because... well again it's not very obvious for someone who reads the document the first time.
```
caddy file-server --listen :8080 --access-log --root /path-to-Cusota-directory
```

closed 2023-08-18 by archandanime