chenxiaolong/Custota #19: Automation?

1-minute-to-midnight, 2023-09-15:
Is it possible to automate everything? Once CalyxOS releases an update, the image is downloaded onto server and signed then hosted on custota automatically? I'm sure there is a neat way to script everything but it would be nice if there was an "official" way to do it.

chenxiaolong, 2023-09-15:
Everything is possible to automate (in that it won't show any interactive prompts), but you'll have to put together a script that fits your use case.

You could maybe start off with something like this:

```bash
# Set up paths and signing key passphrases
DEVICE_CODENAME=cheetah
MAGISK=/path/to/magisk.apk
AVB_KEY=/path/to/avb.key
OTA_KEY=/path/to/ota.key
OTA_CERT=/path/to/ota.crt
export PASSPHRASE_AVB="..."
export PASSPHRASE_OTA="..."

# Find latest version of ROM
URL=$(command to find URL of latest version)
FILENAME=${URL##*/}

if [[ -f "${FILENAME}.patched" ]]; then
    echo "Patched ROM already exists"
    exit
fi

# Download ROM
curl -O "${URL}"

# Patch ROM
avbroot \
    ota patch \
    --input "${FILENAME}" \
    --key-avb "${AVB_KEY}" \
    --key-ota "${OTA_KEY}" \
    --cert-ota "${OTA_CERT}" \
    --magisk "${MAGISK}" \
    --pass-avb-env-var PASSPHRASE_AVB \
    --pass-ota-env-var PASSPHRASE_OTA

# Generate custota csig file
custota-tool \
    gen-csig \
    --input "${FILENAME}.patched" \
    --key "${OTA_KEY}" \
    --cert "${OTA_CERT}" \
    --pass-env-var PASSPHRASE_OTA

# Generate custota update info file
custota-tool \
   gen-update-info \
   --file "${DEVICE_CODENAME}.json" \
   --location "${FILENAME}.patched"

# Copy files to webserver folder (or upload them)
# Eg. $ cp "${FILENAME}.patched" "${FILENAME}.csig" "${DEVICE_CODENAME}.json" /var/www/blah/
# Or  $ rsync "${FILENAME}.patched" "${FILENAME}.csig" "${DEVICE_CODENAME}.json" server:/var/www/blah/
```

@archandanime has another nice example in their repo: https://github.com/archandanime/scripts-and-dotfiles/blob/9dcdfe8540b94e491e8a8c2ebcc1f4da91b80d22/scripts/lynx/patch_n_create_ota.sh

1-minute-to-midnight, 2023-09-15:
I was thinking more like whenever calyxos has an ota update, this script will be triggered and automatically be notified on the phone after the custom ota is made and directly installed. Sort of like doing it without patching from the official source. If this could be achieved I believe more people would try avbroot because it would be as seamless as the official ota after inital setup.

chenxiaolong, 2023-09-15:
I don't think I quite understand what you're describing.

If you're looking to make something that automatically checks for new upstream OTAs, patches them, and uploads the patched OTAs to your server, then the easiest way to do that is by making a script like the above that runs periodically (eg. with a cron job). Once the patched OTAs are hosted on your server, Custota can automatically download and install them.

1-minute-to-midnight, 2023-09-15:
Yeah exactly what you are saying. I'll try doing something like that later. It would be great if you or someone else with time could compile that and put into the official repo. It's a great project btw. This one last thing would be the cherry on top.

chenxiaolong, 2023-09-15:
Sounds good and glad you're liking the project! 

I probably won't end up adding this sort of script to this repo because I like to keep everything as generic as possible and there are way too many possible scenarios to support. That said, if you run into any issues creating a script like this, I'm happy to help.

1-minute-to-midnight, 2023-09-15:
Alright Thanks 😁

closed 2023-09-15 by 1-minute-to-midnight