chenxiaolong/Custota #24: Custota manual is outdated

archandanime, 2023-10-19:
Hello. Thank you for a great app and your efforts to quickly fix things to work with new Android 14!

Today I ran my automatic script to generate a new OTA using `avbroot` and `custota-tool` because there is new KernelSU release :)
The script checks and downloads latest `avbroot` and `custota-tool` binaries, `avbroot` successfully generated the OTA zip but custotal failed to generate `.cig` file:
```bash
error: unrecognized subcommand 'gen-csig'

Usage: custota-tool <COMMAND>

For more information, try '--help'.
error: unrecognized subcommand 'gen-update-info'

Usage: custota-tool <COMMAND>

For more information, try '--help'.
```
I read the [CHANGELOG](https://github.com/chenxiaolong/Custota/blob/v2.4/CHANGELOG.md) but there is no information about how to generate `.csig` file to let Custota app to work(maybe it has been changed?). `custota-tool` command manual doesn't mention it either.

Would you mind updating the manual please? Thank you!

chenxiaolong, 2023-10-19:
Hmm, could you post the output of `custota-tool --help`? The commands haven't changed at all and it should look like this:

```
Usage: custota-tool <COMMAND>

Commands:
  gen-csig         Generate a csig file for an OTA zip
  gen-update-info  Generate or update an update info file
  help             Print this message or the help of the given subcommand(s)

Options:
  -h, --help  Print help
```

Is there any chance `avbroot` somehow got saved as `custota-tool`?

archandanime, 2023-10-19:
Thank you for the quick reply!

This is really embarrassing to me :') You are totally right , my symbolic link for `custota-tool` leads to `avbroot` instead of `custota-tool`, that's why I the command(which actually the symbolic link to `avbroot`) did not show the option:
```bash
> ls -l bin/
total 33248
lrwxrwxrwx 1 squid squid      13 Oct  5 08:09 avbroot -> avbroot_2.2.0
-rwxr-xr-x 1 squid squid 6112960 Sep  7 09:37 avbroot_2.0.1
-rwxr-xr-x 1 squid squid 6256320 Sep  8 13:36 avbroot_2.0.3
-rwxr-xr-x 1 squid squid 7018176 Sep 24 09:42 avbroot_2.1.0
-rwxr-xr-x 1 squid squid 7243456 Oct  5 08:09 avbroot_2.2.0
lrwxrwxrwx 1 squid squid      13 Oct 19 17:07 custota-tool -> avbroot_2.2.0
-rwxr-xr-x 1 squid squid 2406800 Sep  7 09:37 custota-tool_2.2
-rwxr-xr-x 1 squid squid 2492816 Oct  5 08:09 custota-tool_2.3
-rwxr-xr-x 1 squid squid 2492816 Oct 19 17:07 custota-tool_2.4
```
When I executed the symbolic link, it shows that it was `custota-tool` instead of avbroot:
```bash
> bin/custota-tool
Usage: custota-tool <COMMAND>

Commands:
  avb          Pack, unpack, and inspect AVB-protected images
  boot         Pack, unpack, and inspect boot images
  completion   Generate shell tab completion configs
  cpio         Pack, unpack, and inspect cpio archives
  fec          Generate dm-verity FEC data and verify/repair files
  key          Generate and convert keys
  ota          Patch or extract OTA images
  patch        (Deprecated: Use `avbroot ota patch` instead.)
  extract      (Deprecated: Use `avbroot ota extract` instead.)
  magisk-info  (Deprecated: Use `avbroot boot magisk-info` instead.)
  help         Print this message or the help of the given subcommand(s)

Options:
  -h, --help     Print help
  -V, --version  Print version
```
I figured out the binary shows its filename on the command manual instead of its really name:
```bash
> ln -s avbroot this-is-not-avbroot
> ls -l this-is-not-avbroot
lrwxrwxrwx 1 squid squid 7 Oct 19 19:21 this-is-not-avbroot -> avbroot
> ./this-is-not-avbroot
Usage: this-is-not-avbroot <COMMAND>

Commands:
  avb          Pack, unpack, and inspect AVB-protected images
  boot         Pack, unpack, and inspect boot images
  completion   Generate shell tab completion configs
  cpio         Pack, unpack, and inspect cpio archives
  fec          Generate dm-verity FEC data and verify/repair files
  key          Generate and convert keys
  ota          Patch or extract OTA images
  patch        (Deprecated: Use `avbroot ota patch` instead.)
  extract      (Deprecated: Use `avbroot ota extract` instead.)
  magisk-info  (Deprecated: Use `avbroot boot magisk-info` instead.)
  help         Print this message or the help of the given subcommand(s)

Options:
  -h, --help     Print help
  -V, --version  Print version

```
Same with `custota-tool`:
```bash
> ln -s custota-tool_2.4 this-is-not-custota-tool
> ls -l this-is-not-custota-tool
lrwxrwxrwx 1 squid squid 16 Oct 19 19:24 this-is-not-custota-tool -> custota-tool_2.4
> ./this-is-not-custota-tool
Usage: this-is-not-custota-tool <COMMAND>

Commands:
  gen-csig         Generate a csig file for an OTA zip
  gen-update-info  Generate or update an update info file
  help             Print this message or the help of the given subcommand(s)

Options:
  -h, --help  Print help
```

Thank you for the suggestion! I'm closing the issue.

closed 2023-10-19 by archandanime