chenxiaolong/Custota #28: Non-matching update keys

probablypablito, 2023-12-10:
Hey! On my quest to make an automatic update server I've ran into an issue with the Custota app. 

When attempting to check for updates over HTTP (not HTTPS in case it matters) I get the following error: `UpdaterThread$BadFormatException (csig is not signed by a trusted key)`

Despite this issue, I *am* able to manually sideload the __same__ OTA file (at least two files with the same hash) with those same keys. 

Here's the command I am using to generate the csig in case something is desperately wrong:

```python 
command = [
    './binaries/avbroot', 'ota', 'patch',
    '--input', f'./downloads/{device_name}-ota.zip',
    '--key-avb', './keys/avb.key',
    '--key-ota', './keys/ota.key',
    '--cert-ota', './keys/ota.crt',
    '--magisk', './downloads/Magisk.apk',
    '--magisk-preinit-device',  preinit_name,
    '--output', f'./downloads/{device_name}-ota.patched.zip'
]
```

The server logs also show a request for oriole.json and the csig file so that isn't the issue as well. Any help is much appreciated!!

chenxiaolong, 2023-12-10:
If you're getting that error, I think your server is working perfectly!

I'd suggest checking two things:

* Make sure you're using the same `ota.key` for both `avbroot` and `custota-tool`.
* Make sure the [customotacerts Magisk/KernelSU module](https://github.com/chenxiaolong/Custota#custom-verification-key) is installed. This is what installs `ota.crt` as one of Android's allowed verification keys.

probablypablito, 2023-12-10:
Yup! Had the module install but didn't see that I had to do it with my own cert. Totally missed that, thank you.

closed 2023-12-10 by probablypablito