chenxiaolong/Custota #29: IOException (Server does not support byte ranges)

Willie30F, 2023-12-21:
Custota App says:
```
Failed to install OTA update
IOException (Server does not support byte ranges)
```

When I test my server if it supports byte ranges using `curl -I https://p8pro-ota.someserver.de/husky-ota-uq1a.231205.015-c52b44d3.zip`. It outputs:
```
HTTP/1.1 200 OK
Server: nginx
Date: Thu, 21 Dec 2023 20:08:20 GMT
Content-Type: application/zip
Content-Length: 2296984954
...
Accept-Ranges: bytes
```

Another test using `curl -I https://p8pro-ota.someserver.de/husky-ota-uq1a.231205.015-c52b44d3.zip -H "Range: bytes=0-1023"` It outputs:
```
HTTP/1.1 206 Partial Content
Server: nginx
Date: Thu, 21 Dec 2023 20:10:24 GMT
Content-Type: application/zip
Content-Length: 1024
Last-Modified: Thu, 21 Dec 2023 19:07:30 GMT
Connection: keep-alive
...
Content-Range: bytes 0-1023/2296984954
```

Seems to be supported. Any ideas?

chenxiaolong, 2023-12-22:
Hmm, it looks like nginx only reports `Accept-Ranges: bytes` when `Range` wasn't specified in the request.

I'll adjust Custota's logic to check for `Content-Range` in the response instead.

chenxiaolong, 2023-12-22:
This should be fixed by #30. I'll release version 2.5 with that fix in a little bit.

closed 2023-12-22 by chenxiaolong

Willie30F, 2023-12-22:
> This should be fixed by #30. I'll release version 2.5 with that fix in a little bit.

Thanks 🙏. I'll be happy to try it when it's released 👍.

chenxiaolong, 2023-12-22:
Great, thanks for testing!

PR #30 (2023-12-22): UpdaterThread: Check for Content-Range instead of Accept-Ranges
nginx only returns Accept-Ranges if the request was sent without the Ranges header.

Fixes: #29
commit 983158a0 (2023-12-22): UpdaterThread: Check for Content-Range instead of Accept-Ranges

nginx only returns Accept-Ranges if the request was sent without the
Ranges header.

Fixes: #29