chenxiaolong/Custota #37: avbroot signing

NotWellard, 2024-01-15:
Hi, I signed my original ROM with avbroot <3.0.0 but my most recent OTA was signed with 3.0.0 is it possible to use custota without unlocking my bootloader again (necessitating a wipe)? I noticed in the app it shows "no certificates found" so I'm assuming not but I just wanted to make sure before I wiped my phone again due to the inconvenience.

chenxiaolong, 2024-01-15:
Yep, there should never be a need to unlock the bootloader and start over (in general).

Regarding the `no certificates found`, can you double check that you no longer have the `clearotacerts` Magisk module installed? That module does what its name says and is no longer required after avbroot 3.0.0.

NotWellard, 2024-01-15:
Oh right I thought it only cleared the official certs. 🥴Thanks for the quick reply and resolution. I'm very glad I asked now lol.

closed 2024-01-15 by NotWellard