chenxiaolong/Custota #38: Trying to install an update via file path, getting "OS already up to date" message

nad0vs, 2024-02-03:
Basically what it says on the tin can. 
I am reasonably (98.9999%) sure I am not mixing things up and using a new file with a recent OTA freshly cooked via AVBRoot.

Why is this happening and how do I solve it ?

chenxiaolong, 2024-02-04:
Custota will treat the system as being up to date if the Android fingerprint value for the running system matches the file you're trying to flash.

The fingerprint for the running system is shown inside the Custota app and the fingerprint for the OTA is in `META-INF/com/android/metadata` inside the zip (under the `post-build` key).

nad0vs, 2024-02-07:
Hm, okay, so.... suppose I take a fresh CalyxOS OTA from the calyx site, root and resign it with AVBroot

Shouldn't the fingerprint become different? Is there some extra step I need to make to change it?

chenxiaolong, 2024-02-07:
avbroot replaces the signing keys and adds root, but it does not change anything else about the OTA. The fingerprint and all other version numbers remain the same. It's theoretically possible to change the fingerprint, but it's not practical to do so. The fingerprint is embedded into a bunch of different parts of Android and if they don't match, it can cause obscure bugs.

That said, I'm curious why you're running into this issue in the first place. Each CalyxOS update should have a new fingerprint, so updating from an old (patched) build to a new (patched) build shouldn't be a problem. I'd expect you to only run into this if you're trying to "update" to the same CalyxOS version.

NotWellard, 2024-02-15:
I can't speak for nad0vs but I encountered this when a new magisk update was pushed, I couldn't update automatically until the Android version was bumped too.

chenxiaolong, 2024-02-15:
> I can't speak for nad0vs but I encountered this when a new magisk update was pushed, I couldn't update automatically until the Android version was bumped too.

Yep, that's expected behavior for right now. I don't really know of a good way to fix it.

Changing Android's own version numbers is problematic and can easily break things. We might need to resort to having avbroot add a custom custota-specific version field to `build.prop` or something like that.

chenxiaolong, 2024-10-14:
This has been implemented in #92, which will be included in Custota 5.0.

Custota can now detect changes to partition contents, even if the OS version remains the same (as long as the `.csig` file is generated by custota-tool 5.0 too).

commit f4e0f153 (2024-10-14): Use vbmeta digest to detect same-version upgrades

When using Custota in conjunction with avbroot, there may be situations
where a new patched OTA has the same version as the currently running
OS. The most common example is updating Magisk.

With this commit, if the OTA's fingerprint matches the current OS
fingerprint, Custota will compare the vbmeta digest to determine if an
update is necessary. This should reliably detect every scenario where
the OTA's version is the same, but the partition contents differ.

The vbmeta digest is not readily accessible in an OTA, so custota-tool
will compute  [... cut]
commit 8352453f (2024-10-14): Use vbmeta digest to detect same-version upgrades

When using Custota in conjunction with avbroot, there may be situations
where a new patched OTA has the same version as the currently running
OS. The most common example is updating Magisk.

With this commit, if the OTA's fingerprint matches the current OS
fingerprint, Custota will compare the vbmeta digest to determine if an
update is necessary. This should reliably detect every scenario where
the OTA's version is the same, but the partition contents differ.

The vbmeta digest is not readily accessible in an OTA, so custota-tool
will compute  [... cut]
commit 29304c8b (2024-10-14): Use vbmeta digest to detect same-version upgrades

When using Custota in conjunction with avbroot, there may be situations
where a new patched OTA has the same version as the currently running
OS. The most common example is updating Magisk.

With this commit, if the OTA's fingerprint matches the current OS
fingerprint, Custota will compare the vbmeta digest to determine if an
update is necessary. This should reliably detect every scenario where
the OTA's version is the same, but the partition contents differ.

The vbmeta digest is not readily accessible in an OTA, so custota-tool
will compute  [... cut]
PR #92 (2024-10-14): Use vbmeta digest to detect same-version upgrades
When using Custota in conjunction with avbroot, there may be situations where a new patched OTA has the same version as the currently running OS. The most common example is updating Magisk.

With this commit, if the OTA's fingerprint matches the current OS fingerprint, Custota will compare the vbmeta digest to determine if an update is necessary. This should reliably detect every scenario where the OTA's version is the same, but the partition contents differ.

The vbmeta digest is not readily accessible in an OTA, so custota-tool will compute i [... cut]
closed 2024-10-14 by chenxiaolong