chenxiaolong/Custota #39: Crash on update check

probablypablito, 2024-03-03:
I'm on GrapheneOS 14 and connecting over HTTP via a VPN to my update server.

Attached is a log file:
[Custota log 6faa805cb0a6.txt](https://github.com/chenxiaolong/Custota/files/14474336/Custota.log.6faa805cb0a6.txt)

chenxiaolong, 2024-03-03:
That's really strange. It's failing to create a file to save its logs. I'm not sure what would prevent Custota from writing to its own directory. I know GrapheneOS has some storage restrictions, like the storage scopes feature, but that shouldn't apply here.

To try and troubleshoot this, can you post the outputs of these commands?

```
adb shell su -c 'ls -lZd /data/media/0/Android/data/com.chiller3.custota/files /data/data/com.chiller3.custota'
adb shell pm dump com.chiller3.custota | grep 'uid=.*gids='
```

I just want to make sure this isn't just some filesystem permissions issue.

On my device, it looks like this:

```
drwx------ 6 u0_a281 u0_a281     u:object_r:app_data_file:s0:c25,c257,c512,c768       3452 2023-12-12 09:05 /data/data/com.chiller3.custota
                 ^^^     ^^^                              *********************
drwxrws--- 2 u0_a281 ext_data_rw u:object_r:media_rw_data_file:s0:c25,c257,c512,c768  3452 2023-12-30 10:51 /data/media/0/Android/data/com.chiller3.custota/files
                 ^^^                                           *********************
    uid=10281 gids=[] type=0 prot=signature
          ^^^
```

The highlighted parts (user ID `^^^` and SELinux MCS `***`) should match.

probablypablito, 2024-03-03:
```cmd
PS C:\Users\Pablo> adb shell su -c 'ls -lZd /data/media/0/Android/data/com.chiller3.custota/files /data/data/com.chiller3.custota'
drwx------ 6 u0_a77 u0_a77      u:object_r:app_data_file:s0:c77,c256,c512,c768  3452 2024-03-03 14:41 /data/data/com.chiller3.custota
drwxrws--- 2 u0_a77 ext_data_rw u:object_r:media_rw_data_file:s0                3452 2024-02-18 16:49 /data/media/0/Android/data/com.chiller3.custota/files
```

```cmd
PS C:\Users\Pablo> adb shell pm dump com.chiller3.custota | findstr.exe 'uid=.*gids='
    uid=10077 gids=[] type=0 prot=signature
    
```

chenxiaolong, 2024-03-03:
Thanks. That `u:object_r:media_rw_data_file:s0` definitely looks wrong. I would suggest deleting the entire `/data/media/0/Android/data/com.chiller3.custota` directory, rebooting, and seeing if that fixes the problem.

(I'm not sure how it ended up in that state though. That app directory is created by Android before the app launches the first time.)

josevega96, 2024-03-05:
I'm getting a crash also in pixel stock with the march security patch , I could update from the February patch to the march patch without any issue , but after the update I began getting the error 
[image]

chenxiaolong, 2024-03-05:
@josevega96 Thanks. I've opened a new issue over at #40 for this as it's unrelated to this specific issue.

probablypablito, 2024-03-05:
I followed the deletion instructions but now receive this error:

```txt
Failed to install OTA update

IOException (Failed to download update info)
-> SocketException (Binding socket to network
108 failed: EPERM (Operation not permitted)) ->
ErrnoException (Binding socket to network 108
failed: EPERM (Operation not permitted))
```

I've tried re-deleting the directory but to no avail.

chenxiaolong, 2024-03-05:
If I remember correctly, GrapheneOS has a permission toggle for each app for network access. Can you double check that it is enabled?

probablypablito, 2024-03-06:
It is enabled. Do you need more logs or a screen recording? I'm unsure what could've caused this as I had Custota working before I updated to v3 / avbroot v3

probablypablito, 2024-03-06:
```cmd
C:\Users\Pablo>adb shell "ps -efZ | grep custota"
u:r:custota_app:s0:c77,c256,c512,c768 u0_a77 21180  922 11 13:37:26 ?    00:00:00 com.chiller3.custota
u:r:shell:s0                   shell        21220  1249 2 13:37:29 ?     00:00:00 sh -c ps -efZ | grep custota
u:r:shell:s0                   shell        21223 21220 7 13:37:29 ?     00:00:00 grep custota
```

No new output was printed for the second command.

chenxiaolong, 2024-03-06:
Thanks. So everything SELinux-related is good. I really have no idea what else would cause the `ErrnoException (Binding socket to network 108 failed: EPERM (Operation not permitted))`.

The only thing left I could suggest is the brute force approach: maybe try uninstalling the Custota module completely and booting once without it installed (so that Android will delete all of Custota's data). It probably won't help, but I don't really have any more ideas for what the problem might be.

probablypablito, 2024-03-07:
Tried to no avail :(

No idea what it is either, but this is definitely annoying. I'll miss using it! I'll open another issue / contact you if I find a solution.

probablypablito, 2024-03-07:
Oh wow literally just now I found the issue: my WireGuard VPN running in root mode.

WireGuard on Android can be ran either in userspace or (with root) in kernel mode. The latter seems to make Custota unhappy. Turning off the VPN or setting it back to userspace seems to fix that specific issue.

chenxiaolong, 2024-03-07:
Oh wow, nice find.

Custota (and also Android's underlying `update_engine`) use Android's special [`Network` API](https://developer.android.com/reference/android/net/Network) for all connections. It's the standard way to handle things like only connecting if the user is on an unmetered network. I know Android's implementation of this API does a bunch of things behind the scenes, like setting up `fwmark` firewall rules. At least on desktop Linux, WireGuard also sets up its own `fwmark` firewall rules. I wonder that's causing some sort of conflict. Userspace mode wouldn't be affected since that goes through Android's VPN APIs.

nixigaj, 2025-03-16:
Was just about to report an issue here but yeah, I have the same issue with kernel mode WireGuard, even if `AllowedIPs` is only set to a private range that has nothing to do with my OTA server.

chenxiaolong, 2025-03-16:
It might be possible to work around the problem by having Custota pre-download the OTA and then pass a local file to `update_engine`, but I'm not sure. That isn't something I plan to implement though.

chenxiaolong, 2025-03-16:
Actually, I have one final idea. Please give this test build a try: [Custota-5.6.r4.g979cc73-release.zip](https://github.com/user-attachments/files/19273957/Custota-5.6.r4.g979cc73-release.zip)

You'll have to long press Custota's version number to enable the debug options and then turn off the `Pass network ID to update_engine` setting.

I have no idea if this will work. If it does, the main thing that will be impacted is the unmetered network setting. When `Pass network ID to update_engine` is turned off, the unmetered network setting will be respected by Custota, but not by `update_engine`, so it could potentially download the entire OTA over a metered network (eg. if Wi-Fi disconnects during the download).

nixigaj, 2025-03-16:
It did not make a difference unfortunately, same socket exception as before, and works again when disabling the WireGuard interface. If you have time to try more debug builds just send them here and I will try them.

I'm running Android 15 via GrapheneOS 2020030200+ on a Pixel 9 Pro with a self signed custom build with Magisk added, if that is relevant.

chenxiaolong, 2025-03-17:
Thanks for testing! Could you run `adb logcat -s update_engine`, retest one more time with `Pass network ID to update_engine` unchecked, and then post the new log output?

I just want to double check that it's failing for exactly the same reason.

nixigaj, 2025-03-17:
When running `adb logcat -s update_engine` and pressing the `Check for updates` button I get no output at all. I ran just `adb logcat` for 1s while clicking the button and removed 20MB of logs around where I assume the update check event happens. This is the contents of that part:

<details><summary>Logs</summary><pre><code>03-17 04:26:13.679 24943 24943 D UpdaterJob: Scheduling job:  {requiredNetwork=NetworkRequest [ NONE id=0, [ Capabilities: INTERNET&TRUSTED&VALIDATED&NOT_VCN_MANAGED&NOT_BANDWIDTH_CONSTRAINED Uid: 10160 UnderlyingNetworks: Null] ], isRequiredBatteryNotLow=false, isPersisted=false, intervalMillis=0, extras=PersistableBundle[{action=1}]}
03-17 04:26:13.680  1602  3144 D CompatChangeReporter: Compat change id reported: 311402873; UID 10160; state: ENABLED
03-17 04:26:13.680  1602  3144 D CompatChangeReporter: Compat change id reported: 323349338; UID 10160; state: ENABLED
03-17 04:26:13.683 24943 24943 D UpdaterJob: Scheduled job:  {requiredNetwork=NetworkRequest [ NONE id=0, [ Capabilities: INTERNET&TRUSTED&VALIDATED&NOT_VCN_MANAGED&NOT_BANDWIDTH_CONSTRAINED Uid: 10160 UnderlyingNetworks: Null] ], isRequiredBatteryNotLow=false, isPersisted=false, intervalMillis=0, extras=PersistableBundle[{action=1}]}
03-17 04:26:13.698  1602  3144 D CompatChangeReporter: Compat change id reported: 311208749; UID 10160; state: ENABLED
03-17 04:26:13.698  1602  3144 I ActivityManager: Background started FGS: Allowed [callingPackage: com.chiller3.custota; callingUid: 10160; uidState: TOP ; uidBFSL: [BFSL]; intent: Intent { act=com.chiller3.custota.updater.UpdaterService.start cmp=com.chiller3.custota/.updater.UpdaterService (has extras) }; code:PROC_STATE_TOP; tempAllowListReason:<,reasonCode:SYSTEM_ALLOW_LISTED,duration:9223372036854775807,callingUid:-1>; targetSdkVersion:35; callerTargetSdkVersion:35; startForegroundCount:0; bindFromPackage:null: isBindService:false]
03-17 04:26:13.699  1602  3144 D CompatChangeReporter: Compat change id reported: 311208629; UID 10160; state: ENABLED
03-17 04:26:13.705 24943 24943 D UpdaterService: Received intent: Intent { act=com.chiller3.custota.updater.UpdaterService.start cmp=com.chiller3.custota/.updater.UpdaterService (has extras) }
03-17 04:26:13.705 24943 24943 D UpdaterService: Updating foreground notification for state: null
03-17 04:26:13.714  1602 14527 D CompatChangeReporter: Compat change id reported: 296558535; UID 10160; state: ENABLED
03-17 04:26:13.714  1602 14527 D CompatChangeReporter: Compat change id reported: 317799821; UID 10160; state: ENABLED
03-17 04:26:13.714  1602 14527 D CompatChangeReporter: Compat change id reported: 313677553; UID 10160; state: ENABLED
03-17 04:26:13.715  1602 14527 W ForegroundServiceTypeLoggerModule: Foreground service start for UID: 10160 does not have any types
03-17 04:26:13.717 24943 24943 D UpdaterService: Creating new updater thread
03-17 04:26:13.718 24943 24943 D UpdaterService: Updating foreground notification for state: null
03-17 04:26:13.747 24943 24963 D UpdaterThread: onStatusUpdate(0/IDLE, 0.0%)
03-17 04:26:13.747 24943 24943 D UpdaterService: More silent behavior for periodic job: false
03-17 04:26:13.747 24943 24943 D UpdaterService: Trying to stop service if possible
03-17 04:26:13.748 24943 24984 D UpdaterThread: Starting log file (5.6.r4.g979cc73)
03-17 04:26:13.765 24943 24984 D UpdaterThread: Action: CHECK
03-17 04:26:13.765 24943 24984 D UpdaterThread: Waiting for initial engine status
03-17 04:26:13.765 24943 24984 D UpdaterThread: Initial status: 0/IDLE
03-17 04:26:13.765 24943 24984 D UpdaterThread: Starting new update because engine is idle
03-17 04:26:13.766 24943 24984 D UpdaterThread: Update info URI: https://nixigaj.github.io/rooted-graphene/magisk/caiman.json
03-17 04:26:13.766 24943 24943 D UpdaterService: Updating foreground notification for state: ProgressState(type=CHECK, current=0, max=0)
03-17 04:26:13.766 24985 24985 W logcat  : type=1400 audit(0.0:14329): avc:  denied  { read } for  name="u:object_r:userdebug_or_eng_prop:s0" dev="tmpfs" ino=388 scontext=u:r:custota_app:s0:c160,c256,c512,c768 tcontext=u:object_r:userdebug_or_eng_prop:s0 tclass=file permissive=0 app=com.chiller3.custota
03-17 04:26:13.771 24985 24985 W libc    : Access denied finding property "ro.debuggable"
03-17 04:26:13.801  1119  1153 E Netd    : uid 10160 can't select networks other than 31530.
03-17 04:26:13.838  2493  4112 D PowerUI : can't show warning due to - plugged: true status unknown: false
03-17 04:26:13.945  1602  1602 W NotificationHistory: Attempted to add notif for locked/gone/disabled user 0
03-17 04:26:14.015  2493  3005 D LocalImageResolver: Couldn't use ImageDecoder for drawable, falling back to non-resized load.
03-17 04:26:14.037  2493  3005 D LocalImageResolver: Couldn't use ImageDecoder for drawable, falling back to non-resized load.
03-17 04:26:14.048  2493  3005 D LocalImageResolver: Couldn't use ImageDecoder for drawable, falling back to non-resized load.
03-17 04:26:14.082 24943 24984 D TrafficStats: tagSocket(96) with statsTag=0xffffffff, statsUid=-1
03-17 04:26:14.083  1119  1155 E Netd    : uid 10160 can't select networks other than 31530.
03-17 04:26:14.084 24943 24984 D TrafficStats: tagSocket(96) with statsTag=0xffffffff, statsUid=-1
03-17 04:26:14.084  1119  1155 E Netd    : uid 10160 can't select networks other than 31530.
03-17 04:26:14.084 24943 24984 D TrafficStats: tagSocket(96) with statsTag=0xffffffff, statsUid=-1
[... cut ...]
03-17 04:26:15.336  2172  2194 I pixel-thermal: VIRTUAL-USB-UI:35.846 raw data: VIRTUAL-USB-UI:35846 usb_pwr_therm:35846 
03-17 04:26:15.413  1485  1485 D AOC     : A3:MSG: tmd3733_device.cc, 873: USF: TMD3733: Switch lux calc 0->2 d_lux:-.35 m_lux:0.00 R:0.25 C:0.24.
03-17 04:26:15.425  1485  1485 D AOC     : A3:MSG: tmd3733_device.cc, 873: USF: TMD3733: Switch lux calc 2->0 d_lux:5.22 m_lux:0.00 R:0.00 C:0.24.
03-17 04:26:15.801 24943 24997 D ProfileInstaller: Skipping profile installation for com.chiller3.custota</code></pre></details>

chenxiaolong, 2025-03-17:
Oh, that's super interesting. So, the wireguard kernel mode isn't just breaking update_engine. It's also breaking Custota itself (`UpdaterThread` is Custota):
```
03-17 04:26:14.087 24943 24984 E UpdaterThread: Caused by: android.system.ErrnoException: Binding socket to network 104 failed: EPERM (Operation not permitted)
```

Given that, I suspect any app that sets requirements for a specific network type is broken. (It's Android's only reliable API for guaranteeing that traffic doesn't leak to a metered network, but not many apps use it.)

Please give this a try and disable the new `Pin to specific network ID` option: [Custota-5.6.r4.gdcf8001-release.zip](https://github.com/user-attachments/files/19279015/Custota-5.6.r4.gdcf8001-release.zip)

nixigaj, 2025-03-17:
Yep, it works now with `Pin to specific network ID` disabled! It works using both partially and fully tunneled interfaces (`10.x.x.x/24, fdxx::/64` and `0.0.0.0/0, ::/0`). I did an actual GrapheneOS OTA update through a fully tunneled interface for good measure.

Maybe always do the initial update check without pinning to a specific network ID, and then if the actual downloading fails when pinning to a network ID, tell the user that if they use WireGuard in Kernel mode they can try disabling pinning to a specific ID with a warning that it may consume metered network data. Personally I have a beefy data plan so it won't affect me.

chenxiaolong, 2025-03-17:
Great, thanks for testing! I like your suggestions and have implemented them in #126.

closed 2025-03-18 by chenxiaolong

nixigaj, 2025-03-18:
Looks great!

https://github.com/user-attachments/assets/04d84cb3-f93c-4024-a29c-892fa2572208

commit 709b1ddc (2025-03-17): Add option to disable pinning network ID

With how WireGuard's kernel mode sets up the network interfaces, if an
app tries to use network pinning, then netd will block the connection
with an error like:

    uid 10168 can't select networks other than 6896.

On Custota's side, we see -EPERM when binding to the TCP socket.

This commit adds a new debug option for disabling forced network
pinning. When the option is disabled, the unmetered network option is
still guaranteed to be used when scheduling the background jobs.
However, once the job is running, it's only respected on a best effort
basis [... cut]
commit 161a2829 (2025-03-17): Add option to disable pinning network ID

With how WireGuard's kernel mode sets up the network interfaces, if an
app tries to use network pinning, then netd will block the connection
with an error like:

    uid 10168 can't select networks other than 6896.

On Custota's side, we see -EPERM when binding to the TCP socket.

This commit adds a new debug option for disabling forced network
pinning. When the option is disabled, the unmetered network option is
still guaranteed to be used when scheduling the background jobs.
However, once the job is running, it's only respected on a best effort
basis [... cut]
commit fd08a83b (2025-03-17): Add option to disable pinning network ID

With how WireGuard's kernel mode sets up the network interfaces, if an
app tries to use network pinning, then netd will block the connection
with an error like:

    uid 10168 can't select networks other than 6896.

On Custota's side, we see -EPERM when binding to the TCP socket.

This commit adds a new debug option for disabling forced network
pinning. When the option is disabled, the unmetered network option is
still guaranteed to be used when scheduling the background jobs.
However, once the job is running, it's only respected on a best effort
basis [... cut]
commit bf352fdf (2025-03-17): Add option to disable pinning network ID

With how WireGuard's kernel mode sets up the network interfaces, if an
app tries to use network pinning, then netd will block the connection
with an error like:

    uid 10168 can't select networks other than 6896.

On Custota's side, we see -EPERM when binding to the TCP socket.

This commit adds a new debug option for disabling forced network
pinning. When the option is disabled, the unmetered network option is
still guaranteed to be used when scheduling the background jobs.
However, once the job is running, it's only respected on a best effort
basis [... cut]