chenxiaolong/Custota #40: Fails to connect to `update_engine` on Android 14 QPR2

chenxiaolong, 2024-03-05:
Reported here: https://github.com/chenxiaolong/Custota/issues/39#issuecomment-1977697081

chenxiaolong, 2024-03-05:
```
husky:/data/local/tmp # ps -efZ | grep custota
u:r:priv_app:s0:c512,c768      u0_a281      18671  1298 1 20:06:37 ?     00:00:03 com.chiller3.custota
```

Custota isn't running under the `custota_app` SELinux context.

```
husky:/data/local/tmp $ cat custota_selinux.log
----- Environment -----
Timestamp: Mon Mar  4 16:03:32 EST 2024
Script: /data/adb/modules/com.chiller3.custota/post-fs-data.sh
App ID: com.chiller3.custota
App version: v3.0
UID/GID/Context: uid=0(root) gid=0(root) context=u:r:magisk:s0
----- Creating custota_app domain -----
Policy version: 30
----- Updating seapp_contexts -----
SELinux: Loaded file context from:
                /system/etc/selinux/plat_file_contexts
                /system_ext/etc/selinux/system_ext_file_contexts
                /product/etc/selinux/product_file_contexts
                /vendor/etc/selinux/vendor_file_contexts
```

Seems like Android no longer loads `/dev/selinux/apex_seapp_contexts`. The change was [made in AOSP back in September](https://android.googlesource.com/platform/external/selinux/+/e9448817b37b2d14ab8e00dfff4b60347512aae7%5E%21/), but only shipped now in the March 2024 security update.

This kind of sucks. I'll need to see if messing with the [other seapp_contexts files](https://android.googlesource.com/platform/external/selinux/+/refs/tags/android-14.0.0_r29/libselinux/src/android/android_seapp.c#48) is feasible in a Magisk module. If not, we'll need to drop support for Custota's isolated SELinux context, which would be quite unfortunate. (It currently allows Custota to do its job with minimal privileges and avoids granting other system apps access to `update_engine`. Losing this would increase the attack surface quite a bit.)

chenxiaolong, 2024-03-05:
So the `seapp_contexts` file that's edited must match the partition containing the app (`/system` in our case). With `/dev/selinux/apex_seapp_contexts` gone (it was treated as "system"), the only file we can modify now is `/system/etc/selinux/plat_seapp_contexts`.

It's trivial to override any file in `/system` via a Magisk module, but in our case, we need take the original contents and append a line before overriding the file. This *must* be done at boot instead of during module installation because the file must always be up to date. Otherwise, if the file becomes out of date due to an Android update, the risk of a bootloop is high.

All of the SELinux related changes are currently done in `post-fs-data.sh`, which runs before Magisk does its bind mounts, so it hopefully shouldn't be too difficult to switch to modifying `/system/etc/selinux/plat_seapp_contexts`.

closed 2024-03-05 by chenxiaolong

josevega96, 2024-03-05:
Hi, just installed 3.1 and it is still failing to connect, tried getting logs from /storage/emulated/0/Android/data/com.chiller3.custota/files but the folder is empty 
[image]

chenxiaolong, 2024-03-05:
Hmm, can you upload `/data/local/tmp/custota_selinux.log`?

josevega96, 2024-03-05:
Sure, here it is 
[custota_selinux.log](https://github.com/chenxiaolong/Custota/files/14502194/custota_selinux.log)

chenxiaolong, 2024-03-05:
Thanks. That's not the output I expected. Can you upload `/data/adb/modules/com.chiller3.custota/post-fs-data.sh` too?

I think somehow, your device has version 3.1 of the Custota app, but version 3.0 of that script.

josevega96, 2024-03-05:
Sure, here it is 
[Uploading custota_selinux.log…]()

josevega96, 2024-03-05:
You were right even though I updated the app from the kernelsu manager and it said I was using 3.1 it looks like I got a bad download, tried using the version from GitHub and it installed without issue and worked fine, then I tried to downgrade to 3.0 and update from the manger and it worked, the module only doesn't install when I try with the first download kernelsu made , so most likely I just got a bad download. I'm sending the output of the 2 installs 
[image]
[image]

chenxiaolong, 2024-03-06:
Great, glad it's working now!

commit 69a79c08 (2024-03-05): Switch to modifying /system/etc/selinux/plat_seapp_contexts

Android 14 QPR2 (2024 March security update) dropped support for loading
APEX SELinux policies [1] and thus, /dev/selinux/apex_seapp_contexts.

This commit updates the post-fs-data hook script to modify
/system/etc/selinux/plat_seapp_contexts instead, which is the only other
file that's applicable to apps stored on the system partition. The hook
script takes extra care to ensure that an outdated modified version of
this file won't be loaded because that can result in boot loops.

[1] https://android.googlesource.com/platform/external [... cut]
PR #41 (2024-03-05): Switch to modifying /system/etc/selinux/plat_seapp_contexts
Android 14 QPR2 (2024 March security update) dropped support for loading APEX SELinux policies [1] and thus, /dev/selinux/apex_seapp_contexts.

This commit updates the post-fs-data hook script to modify /system/etc/selinux/plat_seapp_contexts instead, which is the only other file that's applicable to apps stored on the system partition. The hook script takes extra care to ensure that an outdated modified version of this file won't be loaded because that can result in boot loops.

[1] https://android.googlesource.com/platform/external/ [... cut]