chenxiaolong/Custota #44: Doesn't work on Pixel 3 Emulator

thearsenaldevteam, 2024-03-12:
When I ran this source code on Pixel 3 Emulator (android 13), the app crashed.
Can you help me check this?
[image]

chenxiaolong, 2024-03-12:
Can you reupload the screenshot? It links back to this page.

thearsenaldevteam, 2024-03-12:
> Can you reupload the screenshot? It links back to this page.

Yes, I updated the screenshot.

chenxiaolong, 2024-03-12:
Ah, it's failing to query the bootloader status. I'm guessing the emulator doesn't implement the bootloader HAL (for querying locked bootloader status, etc.).

Do you know if OTA zips are available for the emulator? If not, I probably won't add a workaround for this since Custota wouldn't be able to install anything even if it could run.

thearsenaldevteam, 2024-03-12:
> Ah, it's failing to query the bootloader status. I'm guessing the emulator doesn't implement the bootloader HAL (for querying locked bootloader status, etc.).
>
> Do you know if OTA zips are available [...]

I'm a newbie to OTA and just saw your project for the first time. I am sorry for my mistake but if you can fix it that's great.

chenxiaolong, 2024-03-12:
Oh, no worries! I looked into it and it looks like Google does not provide OTA updates for emulators. Unfortunately, this means you'll have to use Custota on a real device.

closed 2024-03-12 by chenxiaolong