chenxiaolong/Custota #45: Custoota crashes after rebooting after installation on kernelsu

josevega96, 2024-03-19:
After installing the module in kernelsu it gets installed as com.chiller3.custota.MainAplication it works after adding it in the superuser section, but after rebooting the app doesn't open, to fix it you need to reinstall the apk manually again but after rebooting again it will crash again
[custota.txt](https://github.com/chenxiaolong/Custota/files/14656235/custota.txt)

chenxiaolong, 2024-03-20:
It's crashing because Android is blocking it from accessing `android.os.ServiceManager.getServiceOrThrow`. This means, for whatever reason, Android itself is blocked from accessing the `/system/etc/sysconfig/config-com.chiller3.custota.xml` file from Custota's module.

This is usually caused by root hiding things. Is there anything else you have installed that might be preventing the system from accessing Custota's files? I unfortunately don't really have any other suggestions.

> it gets installed as com.chiller3.custota.MainAplication

This is the same root cause. Same thing if the icon was just a generic icon instead of Custota's icon. It happens when certain Android components (like the home screen) are blocked from accessing Custota's `.apk`.

josevega96, 2024-03-20:
Being doing some testing in a clean installation. the app works on pixel stock february, but once it gets updated to march it stops working after a reboot, also tried installing the module with Magisk and the result is the same

chenxiaolong, 2024-03-21:
Hmm, if it's happening even on a clean installation, I'm not really sure where the issue might be then. I'm personally also using the Pixel stock OS on the March update + Magisk, but I'm not able to reproduce this.