chenxiaolong/Custota #46: Can we remove these 2 lines in AndroidManifest.xml?

thearsenaldevteam, 2024-03-22:
Hi!
After building the .apk file and opening it by APK Editor Studio, I saw these 2 lines exist in the AndroidManifest.xml.
`<uses-library android:name="androidx.window.extensions" android:required="false"/>
<uses-library android:name="androidx.window.sidecar" android:required="false"/>`

I'd like to remove them. Do you know how to resolve it? 
Thank you.

[image]

chenxiaolong, 2024-03-22:
Those come from the official AndroidX libraries. I don't think there's a good way to remove those entries without building those libraries from source (which I have no experience with).

Can I ask why you want to remove the entries? Are there devices where they cause issues?

thearsenaldevteam, 2024-03-23:
Thank you for your response.

_My problem is that if this two lines are inside AndroidManifest.xml of APK, the device compilation doesn´t work. I can remove with this software everytime after compile with android studio, but I want to remove on source code and not include in APK generated by android studio if possible._

chenxiaolong, 2024-03-23:
I'd suggest giving this a try: https://stackoverflow.com/a/31616472. It seems like it might be possible to remove androidx's `uses-library` lines by overriding them in Custota's `AndroidManifest.xml`.

Try adding these lines:

```xml
<uses-library android:name="androidx.window.extensions" android:required="false" tools:node="remove" />
<uses-library android:name="androidx.window.sidecar" android:required="false" tools:node="remove" />
```

thearsenaldevteam, 2024-03-23:
Thank you very much. I'll give a try with your solution.
This issue will be closed.

closed 2024-03-23 by thearsenaldevteam