chenxiaolong/Custota #47: Does this tool have to be used with avbroot?

jeffpeng3, 2024-03-24:
Since the Pixel is no longer capable of OTA updates after rooting, I wanted to try to utilize this tool to accomplish this.
But I've been trying all day and the problem always occurs with the csig signature, but it's impossible for me to get google's private key to sign the csig, which confuses me, does this tool have to be used with avbroot?

chenxiaolong, 2024-03-24:
Custota is meant to be used with OTAs signed by a private key you have. avbroot is one option (which re-signs OTAs and optionally roots them also). Some other folks also use Custota with open-source ROMs (eg. LineageOS) that they compiled themselves.

If you really want to use an unmodified stock Pixel OTA, you can use your own key to sign the `.csig` file, but the corresponding certificate for your key needs to be added into `/system/etc/security/otacerts.zip` (eg. with a Magisk module). I would not recommend this though.

jeffpeng3, 2024-03-25:
Ok, so in my understanding, should csig be signed with the private key of ota too?

Because I was looking at the tool's parameters and noticed that he can accept -c and -C as parameters when generating csig, which seems to be used for csig and ota authentication respectively, and I ran it without any problems, but I got an error when checking for updates.

Maybe I should install the certificate for csig validation somewhere?

Or did I make a mistake somewhere?

```log
03-24 20:36:17.874 17443 26054 D UpdaterThread: Starting new update because engine is idle
03-24 20:36:17.879 17443 17443 D UpdaterService: Updating foreground notification for state: ProgressState(type=INIT, current=0, max=0)
03-24 20:36:17.881 17443 17443 D UpdaterService: Updating foreground notification for state: ProgressState(type=CHECK, current=0, max=0)
03-24 20:36:17.884 17443 26054 D UpdaterThread: Update info URI: content://com.android.externalstorage.documents/tree/primary%3ADownload%2Fota/document/primary%3ADownload%2Fota%2Fhusky.json
03-24 20:36:17.904 17443 26054 D UpdaterThread: Update info: UpdateInfo(version=2, full=LocationInfo(locationOta=husky-ota-ap1a.240305.019.a1-f1130be5.zip, locationCsig=husky-ota-ap1a.240305.019.a1-f1130be5.zip.csig), incremental={})
03-24 20:36:17.910 17443 26054 D UpdaterThread: OTA URI: content://com.android.externalstorage.documents/tree/primary%3ADownload%2Fota/document/primary%3ADownload%2Fota%2Fhusky-ota-ap1a.240305.019.a1-f1130be5.zip
03-24 20:36:17.914 17443 26054 D UpdaterThread: csig URI: content://com.android.externalstorage.documents/tree/primary%3ADownload%2Fota/document/primary%3ADownload%2Fota%2Fhusky-ota-ap1a.240305.019.a1-f1130be5.zip.csig
03-24 20:36:17.933 17443 26054 E UpdaterThread: Failed to install update
03-24 20:36:17.933 17443 26054 E UpdaterThread: com.chiller3.custota.updater.UpdaterThread$BadFormatException: csig is not signed by a trusted key
03-24 20:36:17.933 17443 26054 E UpdaterThread: 	at com.chiller3.custota.updater.UpdaterThread.checkForUpdates(Unknown Source:2138)
03-24 20:36:17.933 17443 26054 E UpdaterThread: 	at com.chiller3.custota.updater.UpdaterThread.run(Unknown Source:259)
03-24 20:36:17.935 17443 26054 D UpdaterThread: Stopping log file
03-24 20:36:17.937 17443 17443 D UpdaterService: Trying to stop service if possible
03-24 20:36:17.937 17443 17443 D UpdaterService: Updater thread completed; stopping service
```

chenxiaolong, 2024-03-25:
Yep, you'll have to grab the existing `/system/etc/security/otacerts.zip` and add your certificate in there for it to be trusted. Note that Android requires the certificate file extension to be `.x509.pem` inside the zip.

The easiest way to replace that zip after it has been updated is by creating a Magisk module.

jeffpeng3, 2024-03-25:
Is this part of the validation done by Custota?

I was wondering if it would be possible to modify some of the source code so that he could skip the validation or specify an external certificate under certain circumstances?

jeffpeng3, 2024-03-25:
If modifying the code makes it work, then I'd like to add a "load trusted 3rd party certificates" feature to the debug options to make it compatible without installing additional modules.

chenxiaolong, 2024-03-25:
Yeah, that's a good idea. I'll work on that today.

chenxiaolong, 2024-03-26:
Could you give #48 a try?

jeffpeng3, 2024-03-26:
It looks like the csig validation has been passed, but I'm experiencing other issues on my device, it looks like it should be similar to what you mentioned [here](https://github.com/chenxiaolong/Custota/issues/45#issuecomment-2008392762).

[check.log](https://github.com/chenxiaolong/Custota/files/14752272/check.log)

This should be due to kernelSU whitelisting, which I fixed by manually extracting the apk and "updating" it.

Now he's working perfectly, and I've successfully OTA'd and switched slots!

It might be possible to provide the apk in the release and mention in the Usage that KernelSU should manually update the app after installing the module and restarting it, so that it works properly.

Or just install a mini apk like magisk and let the user install the full apk the first time they run it, so that every rooting method can follow the same procedure!

jeffpeng3, 2024-03-26:
After dozens of tests, I realized that some of the changes in `post-fs-data.sh` are detected by [banking software](https://play.google.com/store/apps/details?id=com.chinatrust.mobilebank) and denied service...
Since this certificate issue is almost solved, should I open a new issue to solve this problem, or as an alternative solution, I'll just enable this module again every time I need to update it?

chenxiaolong, 2024-03-26:
> Now he's working perfectly, and I've successfully OTA'd and switched slots!

:tada:

I want to find out why KernelSU causes that issue before adding any workarounds. It seems to happen with any module that installs a system app (not just Custota), but I haven't had the time to try and figure out why yet.

> After dozens of tests, I realized that some of the changes in post-fs-data.sh are detected by [banking software](https://play.google.com/store/apps/details?id=com.chinatrust.mobilebank) and denied service...

Darn. I don't try to hide root, so I'm a bit out of the loop with this stuff. If I had to guess, it's probably detecting some of the mount points. Does it work if you comment out this section of `post-fs-data.sh`? https://github.com/chenxiaolong/Custota/blob/132c23c92c3ff6d72c9229fa903a15c2b4d0b34c/app/module/post-fs-data.sh#L40-L64 If you're on the March 2024 update or newer, this should not have much impact. (But on anything older, without this logic, the system only trusts a very small number of certificates for connecting to the OTA server over HTTPS.)

If not, you're probably stuck with enabling/disabling the module when needed. The SELinux parts of `post-fs-data.sh` are required.

jeffpeng3, 2024-03-27:
> I want to find out why KernelSU causes that issue before adding any workarounds. It seems to happen with any module that installs a system app (not just Custota), but I haven't had the time to try and figure out why yet.

Okay, I understand, I'll try to go to the KernelSU community to try to understand the details of this part.

Then I do have the March 2024 update, but after I commented out this section, custota doesn't work properly and is still detected, so it looks like selinux is being detected.

commit 1369eb54 (2024-03-26): Add debug feature for installing custom csig validation certs

This is useful when the csig is signed by a different key than the OTA
and installing the certificate in `/system/etc/security/otacerts.zip`
isn't desired.

Fixes: #47
closed 2024-04-21 by jeffpeng3