chenxiaolong/Custota #50: Proposal: I want to hide all messages and user interaction for use in the background with predefined server.

Finin27, 2024-04-01:
I'm sorry if I bother you., Is possible to contact you? I want to include Custota in our devices (we are a manufacturer) . We need that the app works in the background with our predefined preferences and server, without user interaction. I want to get your approval and (if possible) hire you or some collaborator of this project to modify the code for our target. Thank you so much.

chenxiaolong, 2024-04-02:
Hey Finin27, for that use case, I would suggest using GrapheneOS' system updater as the base instead: https://github.com/GrapheneOS/platform_packages_apps_Updater. Its functionality (support for predefined settings) and license (MIT) is more suited to your use case (Custota is GPLv3).

Finin27, 2024-04-02:
Thank you so much !! I will try with this solution. Also we will try to compile GrapheneOS for our devices. I appreciate tour help. Anything i'm at your disposal. BEST regards from Santander (Spain)!.

closed 2024-04-03 by Finin27