chenxiaolong/Custota #68: Custota app crashes on launch (Android 15 B4)

godspeed0310, 2024-07-20:
The app crashes on the latest version of Android 15, I've attached logs for your reference. [image]

chenxiaolong, 2024-07-20:
Can you upload `/sdcard/Android/data/com.chiller3.custota/files/crash.log`?

Android 15 is not open source yet, so I don't know what changed. However, this looks very much like a bug in the beta where a scheduled job that requests network access is being given a null `Network` instance. If that is the case, there isn't much I can do to work around the problem.

chenxiaolong, 2024-07-20:
I just upgraded by Pixel Tablet to the Android 15 beta and I can't seem to reproduce the problem.

godspeed0310, 2024-07-20:
Here's the log file you asked for
[check.log](https://github.com/user-attachments/files/16322267/check.log)

chenxiaolong, 2024-07-20:
It looks like you uploaded `check.log`. Was there a `crash.log` in that folder?

godspeed0310, 2024-07-21:
Ah yes there was, I'm really sorry. Here's `crash.log`
[crash.log](https://github.com/user-attachments/files/16323149/crash.log)

chenxiaolong, 2024-07-21:
Thanks! I added a workaround in #69. Can you give this test build a try? [Custota-4.6.r2.gf454758-release.zip](https://github.com/user-attachments/files/16325848/Custota-4.6.r2.gf454758-release.zip) After installing, you'll have to disable the `Require Unmetered Network` option in Custota.

Since this is an Android bug, there's not much I can do to fix the actual problem. When the bug occurs, Custota will now show a notification instead of crashing:

[image]

Hopefully Google will fix this before the final Android 15 release.

godspeed0310, 2024-07-23:
Yep the issue seems to be resolved with the build you sent me so I'll close the issue now, tysm for your help

closed 2024-07-23 by godspeed0310

chenxiaolong, 2024-07-23:
Thanks for testing. The PR has been merged and is included in version 4.7.

commit f454758d (2024-07-23): UpdaterJob: Add support for falling back to active network

It appears on that for some folks running the Android 15 beta, Android's
job scheduler implementation is broken and returns a null Network
instance. This commit adds support for falling back to the currently
active network if the Require Unmetered Network option is turned off.
Otherwise, the only thing we can do is show a notification to tell the
user to turn that option off.

Closes: #68
PR #69 (2024-07-23): UpdaterJob: Add support for falling back to active network
It appears on that for some folks running the Android 15 beta, Android's job scheduler implementation is broken and returns a null Network instance. This commit adds support for falling back to the currently active network if the Require Unmetered Network option is turned off. Otherwise, the only thing we can do is show a notification to tell the user to turn that option off.

Closes: #68