chenxiaolong/Custota #72: Is it possible to replace custota-selinux with sesepolicy.rule on KernelSU to avoid being detected

jeffpeng3, 2024-08-05:
I've noticed that a lot of modules that require modifications to the SELinux context use this approach, so I was curious if it could be done in this way and allow the root implementation to hide what might be detected.

sepolicy.rule Reference:
 - [KernelSU Side](https://kernelsu.org/guide/module.html#sepolicy-rule)
 - [Magisk Side](https://topjohnwu.github.io/Magisk/guides.html#sepolicyrule)

chenxiaolong, 2024-08-05:
I don't think that would help. As far as I'm aware, magiskpolicy and custota-selinux both use the exact same approach to modifying the policy.

Custota does not use `sepolicy.rule` files because it's not flexible enough (it cannot copy SELinux types).

chenxiaolong, 2024-08-05:
I think I might know what they're detecting. It's probably the file modification timestamp of `/sys/fs/selinux/load`.

I'm not interested in playing the cat and mouse game for hiding root, but in this case, the workaround should be easy enough. I'll give it a shot after work today.

commit 8096e544 (2024-08-06): custota-selinux: Avoid truncating /sys/fs/selinux/load

This causes the file modification timestamp to change, which some apps
use to detect if the SELinux policy has been modified.

Closes: #72
PR #74 (2024-08-06): custota-selinux: Avoid truncating /sys/fs/selinux/load
This causes the file modification timestamp to change, which some apps use to detect if the SELinux policy has been modified.

Closes: #72
closed 2024-08-06 by chenxiaolong