chenxiaolong/Custota #77: Can Custota be integrated as a system app manually?

Stan805, 2024-08-18:
To start, I would like to apologize in advance if I missed anything in the docs that would answer any of my questions, but does Custota support being installed as a system app natively?

Like if I am modifying a ROM manually on my PC prior to signing it with AVBRoot and flashing, can I edit the system partition priv-apps directory to replace the ROM's original Updater.apk with Custota's APK for a more native/seamless integration?

I would probably have to just do what the startup script that changes SELinux does manually in the rom before I rebuild it. But I am doing this to GrapheneOS, so is there anything else I would have to change?

Is there anything else that is recommended to be done in this case to make it more secure? Will this allow the settings link for system updates to work with Custota? Can I pre-configure Custota with an update server URL?

Also I am aware that rooting isn't recommended, but I think it can be done reasonably securely with Kitsune Mask Magisk, which is a fork of Magisk that allows for better GrapheneOS compatibility and supports an SUList.

chenxiaolong, 2024-08-18:
No worries, always happy to answer any questions!

Integrating Custota directly into the system works fine. This is actually what I do on my personal devices.

> I would probably have to just do what the startup script that changes SELinux does manually in the rom before I rebuild it.

That's correct. You would need to run:

```bash
custota-selinux -s <old policy> <new policy>
```

for `etc/selinux/precompiled_sepolicy` inside `vendor.img` and `sepolicy` inside `vendor_boot.img`. (custota-selinux can run on any Linux system, but not on any other OS.)

`system/etc/selinux/plat_file_contexts` also needs to be updated to include:

```bash
user=_app isPrivApp=true name=com.chiller3.custota domain=custota_app type=app_data_file levelFrom=all
```

> But I am doing this to GrapheneOS, so is there anything else I would have to change?

There isn't anything special to do for GrapheneOS specifically.

> Is there anything else that is recommended to be done in this case to make it more secure?

Nope. Custota in its default configuration already grants itself the least amount of privileges possible. The SELinux changes are also set up to not give other apps any additional permissions.

> Will this allow the settings link for system updates to work with Custota?

Yes, that should automatically work by default.

> Can I pre-configure Custota with an update server URL?

Nope, sorry, this isn't supported.

> Or at this point, is it just better to modify GrapheneOS's system updater to reroute to my update server?

If you're building GrapheneOS from source, I'd definitely just recommend using their updater and changing the URL to point to your server. That would be far easier than unpacking OTAs and filesystems to integrate Custota.

However, if you _are_ interested in using Custota, maybe the script I use for my devices could serve as an example: https://github.com/chenxiaolong/my-avbroot-setup It integrates Custota, along with a few other things.

Stan805, 2024-08-19:
Thank you so much for your response! This will definitely point me more in the right direction since I am new to modifying Android ROMs. I do have one more question though.

If I do chose to use the standard GrapheneOS updater apk, can the web server structure be the same as how Custota's documentation says to make it, or does the GrapheneOS updater require a different structure?

chenxiaolong, 2024-08-19:
No problem!

GrapheneOS' updater uses a very similar structure, but not the same. If I remember correctly, it's 1 text file + the OTA zip itself.

closed 2024-10-08 by chenxiaolong