chenxiaolong/Custota #87: 9/DOWNLOAD_TRANSFER_ERROR

yvay, 2024-10-05:
The update fails on GrapheneOS :(. I suspect a permissions issue (GOS doesn't let system apps access debugging). Here's a grep of "denied" in the app logs:

[custota-denied.txt](https://github.com/user-attachments/files/17265703/custota-denied.txt)

If the issue is not here, please let me know and I'll try uploading a more complete log.

Thank you!

chenxiaolong, 2024-10-05:
Can you share more details about your setup? Specifically, which device you're using and whether you're installing an OTA from the network or from a local directory.

As for the logs, can you post the full output of:
```bash
adb logcat -s update_engine -d
```
immediately after the error occurs?

(All error codes of the form `<number>/<code>` come from Android's `update_engine` and Custota does not have permissions to capture those logs itself.)

yvay, 2024-10-06:
I should have RTFM before opening a useless bug report, sorry.

That log uncovered the real problem: I'm using a self-signed certificate which is rejected by update_engine, but accepted by Custota.

```
[INFO:libcurl_http_fetcher.cc(301)] Setting up curl options for HTTPS
[ERROR:libcurl_http_fetcher.cc(479)] Unable to get http response code.
[ERROR:libcurl_http_fetcher.cc(481)] Return code for the transfer: 60
[INFO:libcurl_http_fetcher.cc(531)] Transfer resulted in an error (0), 3809 bytes downloaded
[INFO:libcurl_http_fetcher.cc(545)] No further proxies, indicating transfer complete
[INFO:multi_range_http_fetcher.cc(170)] Received transfer complete.
[INFO:multi_range_http_fetcher.cc(127)] TransferEnded w/ code 0
[INFO:multi_range_http_fetcher.cc(142)] Didn't get enough bytes. Ending w/ failure.
```

```
# diff -r /apex/com.android.conscrypt/cacerts/ /system/etc/security/cacerts/
Only in /apex/com.android.conscrypt/cacerts/: 2f0b9d1a.0
```

```
# mount | grep cacerts
tmpfs on /apex/com.android.conscrypt/cacerts type tmpfs (rw,seclabel,relatime)
tmpfs on /system/etc/security/cacerts type tmpfs (rw,seclabel,relatime)
worker on /system/etc/security/cacerts type tmpfs (ro,seclabel,relatime,mode=755)
/dev/block/dm-53 on /system/etc/security/cacerts/01419da9.0 type f2fs (ro,lazytime,seclabel,noatime,background_gc=on,discard,no_heap,user_xattr,inline_xattr,acl,inline_data,inline_dentry,flush_merge,extent_cache,mode=adaptive,active_logs=6,reserve_root=32768,resuid=0,resgid=1065,inlinecrypt,alloc_mode=default,checkpoint_merge,fsync_mode=nobarrier,compress_algorithm=lz4,compress_log_size=2,compress_mode=fs,atgc,discard_unit=block,memory=normal)
(...)
```

The top two mounts are from System-CA-certs and they include the custom CA, the others are from Custota and don't include my custom CA. Those are the ones being read by update_engine. Unmounting all but the top two makes the error disappear:

```
# nsenter --mount=/proc/`pgrep update_engine`/ns/mnt -- umount /system/etc/security/cacerts/*
# nsenter --mount=/proc/`pgrep update_engine`/ns/mnt -- umount /system/etc/security/cacerts
```

Apparently, the Custota module copies the certificates from the apex store to the system store to avoid this situation, but ironically in my case it seems to have caused it. Removing this section fixes the issue for me:

```
diff --git a/app/module/post-fs-data.sh b/app/module/post-fs-data.sh
index fe5bd47..1a1bbcf 100644
--- a/app/module/post-fs-data.sh
+++ b/app/module/post-fs-data.sh
@@ -46,38 +46,3 @@ ls -ldZ "${cli_apk%/*}"
 find /data/system/package_cache -name "${app_id}-*" -exec ls -ldZ {} \+

 run_cli_apk com.chiller3.custota.standalone.ClearPackageManagerCachesKt
-
-# Bind mount the appropriate CA stores so that update_engine will use the
-# regular system CA store.
-
-header Linking CA store
-
-apex_store=/apex/com.android.conscrypt/cacerts
-system_store=/system/etc/security/cacerts
-google_store=${system_store}_google
-standard_store=${system_store}
-update_engine_store=${system_store}
-
-if [[ -d "${apex_store}" ]]; then
-    standard_store=${apex_store}
-fi
-
-if [[ -d "${google_store}" ]]; then
-    update_engine_store=${google_store}
-fi
-
-echo "Standard trust store: ${standard_store}"
-echo "update_engine trust store: ${update_engine_store}"
-
-rm -rf "${mod_dir}"/system/etc/security
-
-if [[ "${standard_store}" != "${update_engine_store}" ]]; then
-    mkdir -p "${mod_dir}"/system/etc/security
-    # This copies the directory instead of symlinking it because the SELinux
-    # policy in newer Android versions no longer allows reading the targets of
-    # symlinks labelled with the system_security_cacerts_file type.
-    cp -r "${standard_store}" "${mod_dir}${update_engine_store}"
-
-    # Replace the whole directory instead of merging it.
-    touch "${mod_dir}${update_engine_store}/.replace"
-fi
```

There's one thing I find confusing. My assumption is when two filesystems are mounted at the same path, the one that was mounted later "wins". Therefore, if the Custota module ran before the System-CA-Certs one, I'd expect the bind mount from System-CA-certs to come on top. If System-CA-certs ran first, I'd expect Custota to copy the certificates from the bind mount, including my CA. Neither of those happened. Are Magisk modules loaded in parallel and there's a race condition, or is it a namespace thing where modules don't affect each other while they're being loaded? I'd love to know.

In any case, with the above "fix" I've been able to update to the latest graphene OTA. Thanks again.

chenxiaolong, 2024-10-06:
Thanks for digging into it!

> There's one thing I find confusing. My assumption is when two filesystems are mounted at the same path, the one that was mounted later "wins". Therefore, if the Custota module ran before the [...]

Yeah, that is indeed what should happen.

I don't think it's a race condition. The post-fs-data scripts are executed sequentially ([[1]](https://github.com/topjohnwu/Magisk/blob/7188de12059ba53c6b492a06dd4cddce0c3e368c/native/src/core/module.cpp#L466) -> [[2]](https://github.com/topjohnwu/Magisk/blob/7188de12059ba53c6b492a06dd4cddce0c3e368c/native/src/core/scripting.cpp#L137)), though the load order is arbitrary because it doesn't sort the module names and just depends on filesystem order ([[1]](https://github.com/topjohnwu/Magisk/blob/7188de12059ba53c6b492a06dd4cddce0c3e368c/native/src/core/module.cpp#L386) -> [[2]](https://github.com/topjohnwu/Magisk/blob/7188de12059ba53c6b492a06dd4cddce0c3e368c/native/src/core/module.cpp#L376))..

I'll test this a bit more and see if I can figure out what's going on. I'll probably end up making the Custota post-fs-data script mount things itself (like system-ca-certs) instead of relying on Magisk to do it.

chenxiaolong, 2024-10-06:
Would you mind giving #88 a try? You'll need to flash the new version of Custota and also generate a new system-ca-certs zip with custota-tool.

* Custota app: [Custota-4.11.r2.g8276342-release.zip](https://github.com/user-attachments/files/17271083/Custota-4.11.r2.g8276342-release.zip) (this is signed with the release key, so no need to uninstall the existing version first)
* custota-tool: https://github.com/chenxiaolong/Custota/actions/runs/11205603304?pr=88 (bottom of the page when logged into Github)

---

With this change, on my device, this is what I see when Custota loads before system-ca-certs:

```
tangorpro:/ $ mount | grep cacerts
system_ca_certs on /apex/com.android.conscrypt/cacerts type tmpfs (rw,seclabel,relatime)
system_ca_certs on /system/etc/security/cacerts type tmpfs (rw,seclabel,relatime)

tangorpro:/ $ ls -l {/apex/com.android.conscrypt,/system/etc/security}/cacerts/105c86d9.0
-rw-r--r-- 1 root root 1838 2024-10-06 17:46 /apex/com.android.conscrypt/cacerts/105c86d9.0
-rw-r--r-- 1 root root 1838 2024-10-06 17:46 /system/etc/security/cacerts/105c86d9.0
```

and when system-ca-certs loads before Custota:

```
tangorpro:/ $ mount | grep cacerts
system_ca_certs on /apex/com.android.conscrypt/cacerts type tmpfs (rw,seclabel,relatime,size=3803192k,nr_inodes=950798)
com.chiller3.custota on /system/etc/security/cacerts type tmpfs (rw,seclabel,relatime)

tangorpro:/ $ ls -l {/apex/com.android.conscrypt,/system/etc/security}/cacerts/105c86d9.0
-rw-r--r-- 1 root root 1838 2024-10-06 17:55 /apex/com.android.conscrypt/cacerts/105c86d9.0
-rw-r--r-- 1 root root 1838 2024-10-06 17:55 /system/etc/security/cacerts/105c86d9.0
```

yvay, 2024-10-08:
Thank you! I haven't had much time today but I just updated the two modules and rebooted a couple of times and they seem to be doing good, I get the same output as you. I didn't try installing an OTA but it's pretty obvious it will work.

>Thanks for digging into it!

No problem! I was curious and took it as an opportunity to learn a little about Android and Magisk. Thanks to you for all the helpful explanations and the excellent software.

chenxiaolong, 2024-10-08:
Great, thanks for testing! I'll merge the PR then and release a new version of Custota.

PR #88 (2024-10-08): post-fs-data: Mount cacerts ourselves and unmount shadowed mount points
During early boot, Magisk first runs all `post-fs-data.sh` scripts and then it mounts each module's `<module dir>/system` overrides. This breaks support for using custom CA certs via the system-ca-certs module if it just so happens that filesystem ordering causes Custota to be loaded before system-ca-certs. Custota's gathering of the system certs would run before system-ca-certs, but Magisk's mounting of the resulting files would run after.

To make things easier to troubleshoot, both modules now use the module ID as the a [... cut]
commit 8276342e (2024-10-08): post-fs-data: Mount cacerts ourselves and unmount shadowed mount points

During early boot, Magisk first runs all `post-fs-data.sh` scripts and
then it mounts each module's `<module dir>/system` overrides. This
breaks support for using custom CA certs via the system-ca-certs module
if it just so happens that filesystem ordering causes Custota to be
loaded before system-ca-certs. Custota's gathering of the system certs
would run before system-ca-certs, but Magisk's mounting of the resulting
files would run after.

To make things easier to troubleshoot, both modules now use the module
ID as the  [... cut]
closed 2024-10-08 by chenxiaolong