chenxiaolong/Custota #94: Migtating from csig v1 to v2

schnatterer, 2024-10-17:
Hi @chenxiaolong, 

I'm the autor of [schnatterer/rooted-graphene](https://github.com/schnatterer/rooted-graphene) which relies heavily on custota and zavbroot. Thank you for your great work a d dedication to these projects! 

Could you give me some hints on how to migrate from csig v1 to v2 for a live distro that ships with custota? 

The best idea I came up with would be to release an OTA bundled with custota 5, but create the csig with custota tool 4, in order to allow for seamless OTA updates. 
However, this would work for people that consume every OTA sequentially, but not if an version is missed.

Do you have any better ideas?
Creating two OTAs at different URLs would require users to be aware of the new custota csig version. Sounds not like good UX to me. 

Would it be possible to ship a csig with both versions, that would be compatible with v1?
I read your commit message about strict parsing. So maybe that would be only a possibility for future versions, being less strict and allowing for optional fields.

One thing that could make this whole process a bit easier would be a `- - csig-version` (or similar) parameter that would allow me to keep shipping v1 for some time without having to use different versions of custota and custota tool. Would that be a lot of effort?

Or maybe I'm missing some simpler approach here. Please let me know 🙂

chenxiaolong, 2024-10-18:
Thanks for bringing this up!

I unfortunately did not really do a good job designing the versioning and providing a good upgrade path.

I'm planning to improve this for the future with #96, which disables the strict parsing. From now on, adding new fields to v2 will no longer break backwards compatibility, so I don't expect v3 will ever be necessary (unless Google completely changes how OTAs work).

For now, I think the best path forward is to stick with v1 for a few weeks/months until users upgrade and then switch to v2.

I do like your suggestion for `--csig-version`, so I've implemented that in #95.

schnatterer, 2024-10-18:
Wow, that's awesome! Thanks so much for the fast implementation of my idea🙏

closed 2024-10-18 by schnatterer