chenxiaolong/Custota #103: doesn't work

joyX3, 2024-11-19:
Hello it stops working on the latest system update. I used it for updating to lineage 20241112  it was good but the next one 20241119 can't be installed. I use local source mode. I make 3 files ota.zip.csig, device_name.json and ota.zip. I put them in one folder. I choose this folder as a source in custota.  Then I press check for update button in custota and nothing happens. I don't get any message. I couldn't find any log. I made it with the 5.1 and 5.2 versions

chenxiaolong, 2024-11-19:
Can you reproduce the problem and then immediately grab a logcat with:

```bash
adb logcat -d > logcat.txt
```

Please also grab Android's job scheduler details:

```bash
adb shell dumpsys jobscheduler > jobscheduler.txt
```

(When you press the check for updates button, it submits a scheduled job request to Android, but Android can still decide not to run it. This should hopefully provide hints as to why.)

joyX3, 2024-11-20:
I waited to schelude automatically update message and I got this: 

Background job network restriction are broken in this build of Android. Disable the require unmeterd network options to allow using any active network

I removed cash and data and tried again and get this error:

FileNotFoundException (data/otra_package/care_map.pd: open failed: EACCES (Permission denied) -> ErrnoExceprtion (open failed EACCES (permission denied))

Than I removed cash and data again and reinstalled custota, and it worked

chenxiaolong, 2024-11-20:
Glad it's working for you again. If it ever happens again, please reopen this issue.

> Is it possible make it from the same phone? I don't have access right now

Yep, running as root inside Termux should also work:

```bash
su -c 'logcat -d' > logcat.txt
su -c 'dumpsys jobscheduler' > jobscheduler.txt
```

> Background job network restriction are broken in this build of Android. Disable the require unmeterd network options to allow using any active network

This one is unfortunately an Android bug that was introduced in Android 15 and is still not fixed.

closed 2024-11-20 by chenxiaolong