chenxiaolong/Custota #108: Bootloop on Pixel 9 Pro XL

sepruko, 2024-12-10:
Attempting to update from build `AP3A.241105.008` to `AP4A.241205.013` (both sourced from [here](https://developers.google.com/android/ota#komodo) and patched via `avbroot`) results in bootlooping three times (it just shows the modified OS warning, shows the Google logo then reboots) before reverting to the other slot. I also attempted this with all KSU modules disabled (other than Custota and OEMUnlockOnBoot). I am not willing to test it without a root solution as I do not want to have to wipe the device to re-root it.

<details>
<summary>Full Description</summary>
<br>

1. Check for the new update
2. Install the new update
3. Reboot
4. Show modified OS warning (expected)
5. Show Google 'G' for a moment
6. Repeat steps 3-5 twice more
7. Reboot into other slot

</details>

<details>
<summary>Versions</summary>
<br>

|     Tool     |  Version  |
| :----------: | :-------- |
| avbroot      |   v3.9.0  |
| Custota      |    v5.3   |
| custota-tool |    v5.3   |

</details>

I tried searching online to find methods of locating the boot logs but I came up empty (none of the files indicated existed), so if you could point me in the right direction, that'd be great.

chenxiaolong, 2024-12-10:
Hmm, maybe I'm not looking in the correct place, but from what I can see, KernelSU hasn't released a compatible kernel yet. `AP4A.241205.013` ships with kernel `6.1.84-android14-11-g5c500deec3a7-ab12418271`, but it doesn't look like KernelSU has the corresponding `android14-6.1.84_2024-11` build: https://github.com/tiann/KernelSU/releases/tag/v1.0.2

> I tried searching online to find methods of locating the boot logs but I came up empty (none of the files indicated existed), so if you could point me in the right direction, that'd be great.

I tried intentionally crashing the kernel on my Pixel 9 Pro XL with `echo c > /proc/sysrq-trigger`. It looks like the logs from the last crash are stored in `/sys/fs/pstore/console-ramoops-0`. I'm not sure if these logs get cleared out when it reverts back to the previous slot.

> I am not willing to test it without a root solution as I do not want to have to wipe the device to re-root it.

I don't think it's worth flashing an unrooted build unless you really want to be on the December update sooner. However, it's worth pointing out that switching between rooted and unrooted patched OTAs doesn't require wiping. Sideloading anything signed by your own key from recovery mode will always work as long as it doesn't try to downgrade Android.

sepruko, 2024-12-12:
Hey, thanks for the quick response! Silly me forgot about security patch levels, I guess.

Ended up learning how to build my own Pixel kernel patched with KernelSU (which involved lots of trial, error and guesswork 😛) and everything seems to be in working order, for now.

I'll close this issue since it's everything to do with KernelSU distribution and nothing with Custota.

closed 2024-12-12 by sepruko

chenxiaolong, 2024-12-12:
Nice, glad you got it working with your own kernel!