chenxiaolong/Custota #119: Integrating Custota as a system app

Gontier-Julien, 2025-01-26:
Sorry to bother it kinda a follow up of issue #77 

But integrating Custota as a system app would be like this if i understood right?
And it should just work like that?

```
mkdir ./fs_tree/system/priv-app/com.chiller3.custota/
cp ./Custota-release/system/priv-app/com.chiller3.custota/app-release.apk ./fs_tree/system/priv-app/com.chiller3.custota/com.chiller3.custota.apk

# setting permissions for custota
cp ./Custota-release/system/etc/sysconfig/config-com.chiller3.custota.xml ./fs_tree/system/etc/sysconfig/config-com.chiller3.custota.xml
cp ./Custota-release/system/etc/permissions/privapp-permissions-com.chiller3.custota.xml ./fs_tree/system/etc/permissions/privapp-permissions-com.chiller3.custota.xml

# selinux policy
printf "\nuser=_app isPrivApp=true name=com.chiller3.custota domain=custota_app type=app_data_file levelFrom=all" >> ./fs_tree/system/etc/selinux/plat_seapp_contexts
```

Because i'm kinda confused with the :

>> custota-selinux -s <old policy> <new policy>
> 
> for `etc/selinux/precompiled_sepolicy` inside `vendor.img` and `sepolicy` inside `vendor_boot.img.`

Also thank for the amazing work you rock!

chenxiaolong, 2025-01-26:
> Also thank for the amazing work you rock!

No problem, glad you like it!

> But integrating Custota as a system app would be like this if i understood right?

Yep! That part is all good.

For the `printf` part, it's no longer necessary to hardcode that string. As of Custota 5.2, you can do this instead:

```bash
cat ./Custota-release/plat_seapp_contexts >> ./fs_tree/system/etc/selinux/plat_seapp_contexts
```

> Because i'm kinda confused with the :

The SELinux policy file is stored in two places and can be modified like this:

1. For the `etc/selinux/precompiled_sepolicy` file inside `vendor.img`:

    ```bash
    # Unpack AVB and ext4 filesystem
    avbroot avb unpack -i vendor.img
    afsr unpack -i raw.img

    # Patch SELinux policy file
    custota-selinux -s etc/selinux/precompiled_sepolicy -t etc/selinux/precompiled_sepolicy

    # Pack ext4 filesystem and AVB
    afsr pack -o raw.img
    avbroot avb pack -o vendor.img
    ```

2. For the `sepolicy` file inside the ramdisk of `vendor_boot.img`:

    ```bash
    # Unpack AVB, boot image, and ramdisk
    avbroot avb unpack -i vendor_boot.img
    avbroot boot unpack -i raw.img
    avbroot cpio unpack -i ramdisk.img.0

    # Patch SELinux policy file
    custota-selinux -s cpio_tree/sepolicy -t cpio_tree/sepolicy

    # Repack ramdisk, boot image, and AVB
    avbroot cpio pack -o ramdisk.img.0
    avbroot boot pack -o raw.img
    avbroot avb pack -o vendor_boot.img
    ```

Gontier-Julien, 2025-01-26:
Alright everything work awesome now!
(I did take sometime because of a stupid typo 💃🏻 )

I've also added to both `vendor.img` & `vendor_boot.img` with the` avb.key` i don't know if it needed?

And now the replace part of the ota patch is like this `--replace system extracted/system.img --replace vendor extracted/vendor.img --replace vendor_boot extracted/vendor_boot.img`

It work so it should all be good then right? Unless there is something that i should be aware or that you recommend to do? ^^

chenxiaolong, 2025-01-26:
Great!

> I've also added to both `vendor.img` & `vendor_boot.img` with the` avb.key` i don't know if it needed?

If you're referring to `-k` option in `avbroot avb pack`, then that is completely fine. avbroot will only use the key if it's necessary.

> And now the replace part of the ota patch is like this `--replace system extracted/system.img --replace vendor extracted/vendor.img --replace vendor_boot extracted/vendor_boot.img`
>
> It work so it [...]

Yep, that it! There's shouldn't be anything else you need to do.

Gontier-Julien, 2025-01-26:
Awesome then! Thank you so so so much for the help! You're amazing!!

I can test everything tomorrow when the phone arrive! So i'll keep you updated! But everything should work 🤞🏻 

I'll close this issues then ^^

Gontier-Julien, 2025-01-27:
@chenxiaolong Would you have any idea why the custota app isn't showing?

Installing it manually seem to work, but this isn't really integrated then :/

It is in the /system/priv-app & showing that it there, but when booted it isn't showing anywhere

chenxiaolong, 2025-01-27:
Hmm, the files might have the wrong permissions. What's the output of this?
```bash
ls -lZd /system/priv-app/com.chiller3.custota
ls -lZR /system/priv-app/com.chiller3.custota /system/etc/sysconfig/config-com.chiller3.custota.xml
```

Gontier-Julien, 2025-01-27:
Kinda unrelated to Custota but your magisk module `OEMUnlockOnBoot` i can also extract the apk & add it like Custota?

It would be awesome so i could add it as a safety feature to my own rom for my phone ^^

chenxiaolong, 2025-01-27:
Ah, yep, everything should be `u:object_r:system_file:s0`. That should fix the app not appearing. `vendor_file` is only used for files in `/vendor`.

For `OEMUnlockOnBoot`, you'll need to put the APK somewhere (doesn't matter where) and then you'll need to add an init script to run it like this:

```
# /system/etc/init/oemunlockonboot.rc
service oemunlockonboot /system/bin/app_process / com.chiller3.oemunlockonboot.Main
    oneshot
    class main
    setenv CLASSPATH /system/bin/oemunlockonboot.apk
```

Gontier-Julien, 2025-01-27:
Alright going to test it now!

closed 2025-01-27 by Gontier-Julien

chenxiaolong, 2025-01-27:
Awesome, glad to hear it!