chenxiaolong/Custota #123: Reinstall latest update

schnatterer, 2025-02-09:
Hi @chenxiaolong, 

Do you think it would be possible to add a flag that would force showing the upgrade dialog even if it has the same version as the current one?

The use case I'm thinking about is upgrading to the same version that has some other changes in it (like newer system apps) or switching from rootless to rooted.

Thank you for your fantastic work with custota, avbroot, and others!

chenxiaolong, 2025-02-09:
Glad you like the project!

This is already supported since Custota 5.0 and it doesn't require any user intervention. It now takes both the OS fingerprint and the vbmeta digest into account when determining if an update is available. When you make changes to any of the partitions, the vbmeta digest will change (aside from some esoteric scenarios not found in the wild) and Custota will show that an update is available, even if the OS version remains the same.

(And if you really do want to reinstall the exact same OTA, that's possible too by long pressing Custota's version number to enable debug mode and then enabling the `Allow reinstall` option.)

closed 2025-02-09 by chenxiaolong

schnatterer, 2025-02-09:
Awesome! TIL about debug mode 🥳 thanks for pointing it out! 
Ah, the digest! Likely, upgrading based on the digest didn't work for me because I'm still on csig version 1. Now is the best time to upgrade.