chenxiaolong/Custota #124: Custota 5.6 crashes on checking for update

wonder75, 2025-02-09:
Hi, I updated to Release 5.6. Unsure which version I used before. Magisk somehow disabled the module by itself. I realized because I was getting no updates anymore. So I checked Magisk, reeanbled Custota and updated to latest.

Now I can't check for updates anymore.
When I open the app and click on "check for updates" it closes instantly.

When I do this twice I get a message from Android, that the app has crashed.

Device: Pixel 6a
OS: GOS

Crashlog:

`
type: crash
osVersion: google/bluejay/bluejay:15/AP4A.250105.002/2025011500:user/release-keys
flags: dev options enabled
package: com.chiller3.custota:329216, targetSdk 35
process: com.chiller3.custota
processUptime: 3888 + 755 ms

java.io.IOException: Cannot run program "logcat": /storage/emulated/0/Android/data/com.chiller3.custota/files/check.log: open failed: EACCES (Permission denied)
    at java.lang.ProcessBuilder.start(ProcessBuilder.java:1050)
    at com.chiller3.custota.updater.UpdaterThread.run(Unknown Source:78)
Caused by: java.io.FileNotFoundException: /storage/emulated/0/Android/data/com.chiller3.custota/files/check.log: open failed: EACCES (Permission denied)
    at libcore.io.IoBridge.open(IoBridge.java:583)
    at java.io.FileOutputStream.<init>(FileOutputStream.java:259)
    at java.lang.ProcessImpl.start(ProcessImpl.java:118)
    at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
    ... 1 more
Caused by: android.system.ErrnoException: open failed: EACCES (Permission denied)
    at libcore.io.Linux.open(Native Method)
    at libcore.io.ForwardingOs.open(ForwardingOs.java:563)
    at libcore.io.BlockGuardOs.open(BlockGuardOs.java:274)
    at libcore.io.ForwardingOs.open(ForwardingOs.java:563)
    at android.app.ActivityThread$AndroidOs.open(ActivityThread.java:8712)
    at libcore.io.IoBridge.open(IoBridge.java:568)
    ... 4 more
`

wonder75, 2025-02-09:
OK, I deleted /storage/emulated/0/Android/data/com.chiller3.custota/files/check.log and the other logfiles in that directory.
Afterwards I was able to check for updates. Update is installing no, will post here if it works.

wonder75, 2025-02-09:
The update downloaded and installed fine now. I guess this had nothing to do with 5.6. Somehow the logfiles must have gotten permissions, that were not accessible anymore by Custota. I never manualy changed any permission on my device, so I have no idea how this happened. Maybe it's a problem with Magisk, because the module got somewhow auto-disabled as well.

If this happens to somebody else, deleting the files in
/storage/emulated/0/Android/data/com.chiller3.custota/files/
worked for me.

Guess this is resolved for me.
Good chance, that there is no direct connection to Cutota concerning this problem. Somehow the logfiles got bad permissions and Custota was unable to read them. Sadly i forgot to check the permissions of the files before deleting them. Will do if it happens again.

chenxiaolong, 2025-02-09:
> Magisk somehow disabled the 5.3 module by itself

I'm not sure what would have caused Magisk to do this, but this is indeed the root cause. When an app suddenly disappears, Android automatically deletes its internal data directory (`/data/data/com.chiller3.custota`), but it does not delete the external data directory (`/sdcard/Android/data/com.chiller3.custota`). When the app is installed again, it'll likely get assigned a new user ID, making it unable to access the old/existing external data directory.

> If this happens to somebody else, deleting the files in
> /storage/emulated/0/Android/data/com.chiller3.custota/files/
> worked for me.

Great! This is the proper workaround for the problem.

closed 2025-02-09 by chenxiaolong