chenxiaolong/Custota #129: Patched GKI causes an error

Vagelis1608, 2025-05-22:
Hello, when I try to run an update with Custota from a local dir, I get an error that multiple fingerprints exist in the payload.bin
Well, I actually get a notification about the network id pinning, but the logcat actually tells me whats going on. The debug option that disables network pinning doesn't help, of course. The wrong notification is also a (separate) problem.
I swapped out the kernel image in the boot partition for a GKI that includes support for KernelSU and SuSFS( previous update I used the original KernelSU, this time I used KernelSU-Next ) if it matters.
AVBRoot threw no errors when I patched the zip, adb sideload worked, and everything boots fine, btw.
I even went through the closed issues to see if I could find something similar.

`adb logcat -s UpdaterThread` output:

> ...
05-22 02:55:45.850 23378 26080 D UpdaterThread: OTA metadata: # android.ota.OtaPackageMetadata$OtaMetadata@65504b48
05-22 02:55:45.850 23378 26080 D UpdaterThread: postcondition {
05-22 02:55:45.850 23378 26080 D UpdaterThread:   build_incremental: "2505061805"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   build: "Nothing/Pong/Pong:15/AQ3A.240929.001/2505061805:user/release-keys"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   build: "Nothing/PongEEA/Pong:15/AQ3A.240929.001/2505061805:user/release-keys"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   build: "Nothing/PongIND/Pong:15/AQ3A.240929.001/2505061805:user/release-keys"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   device: "Pong"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   partition_state {
05-22 02:55:45.850 23378 26080 D UpdaterThread:     build: "Android/gsi_arm64/generic_arm64:12/SGR1.240502.001.B6/12299972:user/release-keys"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     device: "generic_arm64"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     partition_name: "boot"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     version: "1725003771"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   }
05-22 02:55:45.850 23378 26080 D UpdaterThread:   partition_state {
05-22 02:55:45.850 23378 26080 D UpdaterThread:     build: "Nothing/Pong/Pong:12/SKQ1.240903.001/2505061805:user/release-keys"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     device: "Pong"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     partition_name: "odm"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     version: "1746525904"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   }
05-22 02:55:45.850 23378 26080 D UpdaterThread:   partition_state {
05-22 02:55:45.850 23378 26080 D UpdaterThread:     build: "Nothing/qssi/qssi:15/AQ3A.240929.001/2505061805:user/release-keys"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     device: "qssi"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     partition_name: "product"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     version: "1746525904"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   }
05-22 02:55:45.850 23378 26080 D UpdaterThread:   partition_state {
05-22 02:55:45.850 23378 26080 D UpdaterThread:     build: "Nothing/qssi/qssi:15/AQ3A.240929.001/2505061805:user/release-keys"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     device: "qssi"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     partition_name: "system"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     version: "1746525904"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   }
05-22 02:55:45.850 23378 26080 D UpdaterThread:   partition_state {
05-22 02:55:45.850 23378 26080 D UpdaterThread:     build: "Nothing/qssi/qssi:15/AQ3A.240929.001/2505061805:user/release-keys"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     device: "qssi"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     partition_name: "system_ext"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     version: "1746525904"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   }
05-22 02:55:45.850 23378 26080 D UpdaterThread:   partition_state {
05-22 02:55:45.850 23378 26080 D UpdaterThread:     build: "Nothing/Pong/Pong:12/SKQ1.240903.001/2505061805:user/release-keys"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     device: "Pong"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     partition_name: "vendor"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     version: "1746525904"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   }
05-22 02:55:45.850 23378 26080 D UpdaterThread:   partition_state {
05-22 02:55:45.850 23378 26080 D UpdaterThread:     build: "Nothing/Pong/Pong:12/SKQ1.240903.001/2505061805:user/release-keys"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     device: "Pong"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     partition_name: "vendor_dlkm"
05-22 02:55:45.850 23378 26080 D UpdaterThread:     version: "1746525904"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   }
05-22 02:55:45.850 23378 26080 D UpdaterThread:   sdk_level: "35"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   security_patch_level: "2025-05-01"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   timestamp: 1746525904
05-22 02:55:45.850 23378 26080 D UpdaterThread: }
05-22 02:55:45.850 23378 26080 D UpdaterThread: precondition {
05-22 02:55:45.850 23378 26080 D UpdaterThread:   device: "Pong"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   timestamp: 0
05-22 02:55:45.850 23378 26080 D UpdaterThread: }
05-22 02:55:45.850 23378 26080 D UpdaterThread: property_files {
05-22 02:55:45.850 23378 26080 D UpdaterThread:   key: "ota-property-files"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   value: "payload_metadata.bin:3979:209432,payload.bin:3979:3488888360,payload_properties.txt:3488892407:156,apex_info.pb:1881:1263,care_map.pb:3201:721,metadata:3488892638:786,metadata.pb:3488893502:1373 "
05-22 02:55:45.850 23378 26080 D UpdaterThread: }
05-22 02:55:45.850 23378 26080 D UpdaterThread: property_files {
05-22 02:55:45.850 23378 26080 D UpdaterThread:   key: "ota-streaming-property-files"
05-22 02:55:45.850 23378 26080 D UpdaterThread:   value: "payload.bin:3979:3488888360,payload_properties.txt:3488892407:156,apex_info.pb:1881:1263,care_map.pb:3201:721,metadata:3488892638:786,metadata.pb:3488893502:1373 "
05-22 02:55:45.850 23378 26080 D UpdaterThread: }
05-22 02:55:45.850 23378 26080 D UpdaterThread: required_cache: 0
05-22 02:55:45.850 23378 26080 D UpdaterThread: type: AB
05-22 02:55:45.850 23378 26080 D UpdaterThread: type_value: 1
05-22 02:55:45.850 23378 26080 D UpdaterThread: Delete /data/ota_package/care_map.pb result: false
05-22 02:55:45.850 23378 26080 E UpdaterThread: Failed to install update
05-22 02:55:45.850 23378 26080 E UpdaterThread: com.chiller3.custota.updater.UpdaterThread$BadFormatException: Metadata postcondition lists multiple fingerprints
05-22 02:55:45.850 23378 26080 E UpdaterThread:         at com.chiller3.custota.updater.UpdaterThread.checkForUpdates(Unknown Source:2205)
05-22 02:55:45.850 23378 26080 E UpdaterThread:         at com.chiller3.custota.updater.UpdaterThread.run(Unknown Source:283)
05-22 02:55:45.851 23378 26080 D UpdaterThread: Stopping log file

chenxiaolong, 2025-05-22:
Interesting. Any chance you could post a link to the original OTA?

Multiple fingerprints in the OTA preconditions is normal and expected, especially for incremental OTAs that support upgrading from multiple old versions. However, in your case, there are multiple fingerprints in the OTA postconditions, which makes no sense. After installation, Android only has 1 fingerprint.

I suspect Nothing didn't intend to build the OTA this way, but it happens to work fine in recovery (which doesn't check the fingerprint) and their OTA updater app. Anyway, I'll add support for this.

Vagelis1608, 2025-05-22:
Well, to be fair, they technically don't provide full OTAs for all their updates. 
I emailed them asking for this one, and they send me a GDrive link. 
Not sure if there would be a problem with me sharing it on the open internet.

chenxiaolong, 2025-05-22:
I downloaded one of the full OTAs from https://github.com/spike0en/nothing_archive and it looks like you're right.

`/odm/etc/build.prop` has:
```
import /odm/etc/build_${ro.boot.hardware.sku}.prop
```

The "default" fingerprint is in `/system/build.prop` and then they conditionally load `/odm/etc/build_EEA.prop` or `/odm/etc/build_IND.prop` for certain regions.

chenxiaolong, 2025-05-22:
This is fixed in #130 and will be included in Custota 5.8. I plan to release that later tonight.

closed 2025-05-22 by chenxiaolong

Vagelis1608, 2025-05-22:
I downloaded the previous release from there too. 
Same behavior, but I didn't bother looking into it back then, figured it might have been an one-off. 
But now I figured I should look into it. 
Can't say that I've seen this before, but honestly, I'm surprised more OEMs aren't doing it. 

Anyway, glad to see the fix was relatively easy. 
I'll download your new release tomorrow, to confirm it's good. 

Thank you.

Vagelis1608, 2025-05-22:
Confirmed no errors, notification says Up To Date
Thanks for the quick fix, and have a good day!

PR #130 (2025-05-22): Allow OTAs with multiple fingerprints in postconditions
Some Nothing devices ship firmware for multiple regions in the same full OTA. These have the default fingerprint in `/system/build.prop` and then conditionally load region-specific propery files from `/odm/etc`. The OTA metadata lists all of these fingerprints in the postconditions.

Fixes: #129
commit 11d28ecc (2025-05-22): Allow OTAs with multiple fingerprints in postconditions

Some Nothing devices ship firmware for multiple regions in the same full
OTA. These have the default fingerprint in /system/build.prop and then
conditionally load region-specific propery files from /odm/etc. The OTA
metadata lists all of these fingerprints in the postconditions.

Fixes: #129