chenxiaolong/Custota #139: App crashes when checking for updates on GrapheneOS 2025070100 (Android 16 alpha)

alphapuggle, 2025-07-03:
I've installed the alpha version of GrapheneOS which is now based on Android 16 using custota. Trying to install an update now causes custota to crash and error about permissions to access logcat. I'm unsure if this is a new GrapheneOS limitation or an issue with Android 16.

I recognize that this is a totally unsupported configuration, but since this is early days for Android 16 I figure the data may help if this is a bigger issue.

Occurs on both 5.11 and 5.10, which I had been using before the upgrade and downgraded to for testing.

[crash.log](https://github.com/user-attachments/files/21040066/crash.log)

chenxiaolong, 2025-07-03:
Great! Yeah, it's almost certainly a Magisk bug. I don't think there's anything I can do to prevent it from happening in the first place, but I might be able to add a boot script that automatically deletes that directory when needed. It only stores log files anyway.

chenxiaolong, 2025-07-11:
I played around with boot scripts and couldn't find a good way to have it run at the appropriate time. It's the only component with root access (the app doesn't use root) but it runs too early, before the user unlocks the device and decrypts the storage. I think for now, I'll just leave this as something that requires a manual workaround.

I do find it weird that it has an app ID of 10000. That's the very first app ID available to apps and Android generally just keeps incrementing the number for each newly installed app. I wonder if the Android 16 upgrade reset the counter and somehow happened to cause this issue.

closed 2025-07-11 by chenxiaolong